gergelytakacs/AutomationShield #357: AeroShield

benodomi, 2023-09-13:
When setting macro #define MANUAL in AeroShield_EMPC.ino 1 I've got an compilation because r has no type.

benodomi, 2023-09-13:
After turning with knob reference doesn't change. Adding a line Xr[1] = r;  on line 104 fixs this.

gergelytakacs, 2023-09-14:
Hello @benodomi and we appreciate the feedback. @erik1392 this is a quick fix - can you please implement it? I don't want to do it without testing.

erik1392, 2023-09-14:
"I'm aware of the bug. There are more like it. The AeroShield_eMPC branch is still under development, hence, it has not yet been merged into the master branch."

benodomi, 2023-09-14:
I've already used this code during Workshop and project work. Sorry for this way of sharing code but this piece of code is with all fixes related to this issue:

```
  #if MANUAL
    float r =  AutomationShield.mapFloat(AeroShield.referenceRead(),0,100,0,M_PI);          //--Sensing Pot reference
    Xr[1] = r;        
  #elif !MANUAL
    if(i >= sizeof(R)/sizeof(float)){ // If trajectory ended
      AeroShield.actuatorWriteVolt(0.0); // Stop the Motor
      while(true); // End of program execution
    }
    if (k % (T*i) == 0){ // Moving through trajectory values    
    //r = R[i];
    Xr[1] = R[i];
      i++;             // Change input value after defined amount of samples
    }
    k++;                              // Increment
  #endif
```

erik1392, 2023-09-14:
Thank you, I will integrate your code.

closed 2023-09-19 by erik1392